libhal/SJSU-Dev2 #5: SPI Driver

kammce, 2018-04-09:


diarmadhi, 2018-06-20:
For the Integration testing section on the design doc, you made a comment about what I wrote is more application then proving it works on hardware. Then you comment that the Demonstration section you say that is were the application goes. 

So is it acceptable to move what's in integration to demonstration? I also did not think we'd being doing source code with the design doc, so hadn't included any previously.

diarmadhi, 2018-06-20:
Also, most of what I can think of as "caveats" seem to fall under you comments about design choices. I'm not sure what would constitute a caveat in this driver.

kammce, 2018-06-21:
These comments are not appropriate for this forum.

closed 2019-06-27 by kammce