libhal/SJSU-Dev2 #245: Potentially removing invalid constexpr constructors

jpan127, 2018-11-07:
Would like a discussion on this.

Most if not all the classes in this repository have constructors with a constexpr specifier.  This is perfectly valid C++ code.  However, there is a flaw in the methodology because most if not all of the classes will never be used in a constexpr context because their methods are non-constexpr.  In my opinion this invalidates the purpose of a `constexpr constructor`.

kammce, 2018-11-07:
`constexpr constructor` are simply constructed at compile time. So if you dumped the `.data` or `.rodata`, you would see that your object was initialized to that state. The just eliminates the need to run the constructor at start up. 

In cases where the constructor cannot be run at compile time, for example a non-static local variable of function, then the constructor becomes real and is put into the `.text` section of the code. But only those local/stack objects will have to run the constructor. All global objects will have had their constructors run at compile. This doesn't matter that much when all you are doing is setting up local variables, but on particular cases, for example the SD card driver (not out yet), we have to calculate a CRC lookup table. This can be done within the `contexpr constructor` rather than at runtime.

jpan127, 2018-11-07:
If I understand you you are saying non-constexpr static / global variables will call their constexpr constructor on construction?

```c++
struct Register {
    constexpr Register(const uint32_t v) : value(v){}
    void set(uint32_t v) { value = v; }
    uint32_t get() { return value; }
    uint32_t value;
};

Register r1{5}; /// Global, will call constructor at compile time?
```

My understanding is that `r1` above will not call its constexpr constructor.  Only a constexpr declaration can call a constexpr constructor, otherwise it will default to a non-constexpr context.

```c++
constexpr Register r2{6};
```

kammce, 2018-11-08:
Try running this on your machine. You will see that the constructor does not appear in the symbol table if the compiler can run the constructor at compile time. This happens even at optimization level 0.

To answer your question, you are correct. The constexpr constructor will be called and you need not make your class constexpr for it to happen.

https://gist.github.com/kammce/d0c3509afeb51fc1d3b34707f385bf33

kammce, 2018-11-08:
Updated the gist with some new findings.

jpan127, 2018-11-08:
Interesting, I did not know it removed it from the symbol table.

To explore further, I let godbolt to disassemble.  It looks like the construction handling depends on the scenario.

- `constexpr` constructor + non-constexpr global = compile time into the `.global` section
- `constexpr` constructor + constexpr global = compile time into the `.global` section
- `constexpr` constructor + non-constexpr local = inlined constructor
- `constexpr` constructor + constexpr local = inlined constructor
- `constexpr` constructor + static non-constexpr local = compile time into `.rodata`
- `constexpr` constructor + static constexpr local = compile time, no section?

- `non-constexpr` constructor + global = function
- `non-constexpr` constructor + local = function
- `non-constexpr` constructor + static local = function

The scenario I'm most interested in is `constexpr constructor and non-constexpr local`.
It is interesting that it is inlined, I thought it would just default to a function.

Also, `constexpr constructor and constexpr local` is also inlined.
It is true the constexpr specifier is merely a suggestion, although its main purpose is really
to check for UB and the possibility of static initialization.

Link : https://godbolt.org/z/R-NCNU

kammce, 2018-11-08:
Yup, yup. I think one of the biggest issues with using constexpr is the fact that it is hard to know if its being inlined, has been converted into a function or is actually used in compile time. `static_assert` can help at times, but more times then not, is not the right solution.

As for why made this a standard for most drivers and classes in SJSU-Dev2, its basically to take advantage compile time initialization of objects that would be in the .data section anyway. If the developer using this frameworks, does their due diligence to place the objects in the appropriate locations, then they can see a performance boost.

With that I am closing this discussion. If you would like to continue the conversation, you can join the slack workspace. Click on the badge on the readme.

closed 2018-11-08 by kammce