libhal/SJSU-Dev2 #267: Move away from using config::kCpuSpeed -> system_clock.GetFrequency()

kammce, 2018-12-06:
system_clock.GetFrequency() will return the actual cpu speed. In the event that the cpu speed is changed at runtime, using the constexpr clock speed is not always a good idea. Moving drivers to use this will make sure that clock speeds are where they should be throughout the runtime of the code.

rabeelio, 2018-12-10:
This is kind of related. I noticed that the LPC40xx's peripheral bus shares the same clock unlike the LPC17xx where each controller has its own configurable clock. I also found out that the current ADC driver changes the peripheral clock bus speed and other drivers may be unaware of this. CAN, for example, derives it's own bus clock from the peripheral bus clock. In this case, we might need a getter for peripheral bus clock frequency. However, it's still possible for a user to init drivers in a random order without knowing that one may be changing the peripheral bus clock.

kammce, 2018-12-13:
Yes, I saw this too. One thing we are going to have to do is come up with a set of clear rules about designing drivers for SJSU-Dev2. Temporally, at the time in which the ADC driver was created, there was no code handling the clockspeed of the peripheral bus, so that engineer set it to what they needed. We should definitely come up with a standard about this and ways to handle different clock speeds.

I should be coming up with a PR soon that should help with this.

kammce, 2019-01-02:
Should be resolved by #293

closed 2019-01-05 by kammce