libhal/SJSU-Dev2 #324: Parallel Bus HAL

kammce, 2019-01-16:
Drivers like stu7060 build their own structure full of GPIOs to handle manipulating a bus full of gpios. I think it would be fitting to abstract that out of the stu7060 class and make a generic Parallel Bus class. The will allow the easy creation of other customized bus types for other future drivers. 

This abstraction should also benefit of making compiling the stu7060 class test file faster. The parallel test should have fewer test cases, making the compilation time smaller. The stu7060 will simply be using an interface to a parallel bus object which should be lightning fast to compile compared to the large mocked structure for the current test.

kammce, 2019-04-28:
Should take a std::array<Gpio>, initializer_list of Gpio objects (if possible), or an array with length.

brandonneep, 2019-06-26:
I'm on it

commit 8ce79200 (2019-09-09): Add ParallelBus library + ParallelGpio

Resolves #324
commit 2c27478f (2019-09-10): Add ParallelBus library + ParallelGpio

Resolves #324
commit fb994642 (2019-09-10): Add ParallelBus library + ParallelGpio

Resolves #324
commit ae0dfa70 (2019-09-12): Add ParallelBus library + ParallelGpio

Resolves #324