libhal/SJSU-Dev2 #392: Discrepancy between time units

harmandersihra, 2019-03-06:
https://github.com/kammce/SJSU-Dev2/blob/3acc8e268ac9e30259a7da7aa33fbb7db4dabecc/firmware/library/utility/time.hpp#L32

Uptime() supposedly returns time in nanoseconds, but Milliseconds returns milliseconds by dividing nanoseconds by 1,000 when it should be by 1,000,000. Uptime() seems like it's actually returning microseconds and not nanoseconds.

kammce, 2019-03-11:
Whoops. That should be changed. At the moment, we are only getting the time in microseconds. So we should change the API to state that the time in microseconds and not nanoseconds.

kammce, 2019-03-11:
@harmandersihra can you make that change and create a PR.

kammce, 2019-03-24:
Resolved by #393

closed 2019-03-24 by kammce