libhal/SJSU-Dev2 #444: Large delays in SD Card Driver

granthaack, 2019-04-27:
In the SD card driver at /library/L2_HAL/memory/sd.hpp there are very large hardcoded delays in the `ReadBlock()` function. These delays are passed in as parameters to the `SendCmd()` function. For example, line 699:

`SendCmd(read_cmd, address, sd.response.data.byte, 100, KeepAlive::kYes); //100ms delay`

There are a couple more examples peppered throughout the function. I've been able to shorten all of these down to around ~10ms and still get reliable SD card operation.

This isn't a bug, but it will make the MP3 project near impossible to complete. Data cannot be read fast enough from the SD card when the delays are in place, the song just starts and stops.

ntwong0, 2019-06-12:
@kammce checking assignment

rjavier441, 2019-06-29:
On It.

commit 1d992998 (2019-08-23): Fix typo which causes SD card status responses to fail

Removes delay times within SD Driver's internal SendCmd() calls.
Removed SD interface, placed virtual methods as part of SD class,
and separated CRC logic into utility file crc.hpp

Fix lint, tidy, and header inclusion issues for crc.hpp

Fix crc.hpp not including type_traits

Resolves #444
closed 2019-08-23 by kammce