libhal/SJSU-Dev2 #529: I/O Expander: PCA9535

kammce, 2019-06-12:


ntwong0, 2019-06-20:
Assign me please

kammce, 2019-07-03:
Something to bear in mind with the L2 project is that it actually implements L1 pwm.hpp. This driver should be able to be used in place of a L1 peripheral pwm if the developer desires.

ntwong0, 2019-08-05:
> Something to bear in mind with the L2 project is that it actually implements L1 pwm.hpp. This driver should be able to be used in place of a L1 peripheral pwm if the developer desires.

i.e. to control brightness on the seven segment, or is this a generalization for PCA9535s that would be off-board?

ntwong0, 2019-08-20:
@kammce is there a better place to do this than in `main.cpp`?

https://github.com/ntwong0/SJSU-Dev2/blob/56112f0b5578aa4f1ab173327c6a8caeb65c5b48/demos/sjone/pca9535/source/main.cpp#L16-L20

kammce, 2019-08-20:
> @kammce is there a better place to do this than in `main.cpp`?
> 
> https://github.com/ntwong0/SJSU-Dev2/blob/56112f0b5578aa4f1ab173327c6a8caeb65c5b48/demos/sjone/pca9535/source/main.cpp#L16-L20

If you are testing that is fine. You would typically copy the `starter` and work in there if you were testing and/or making a demo. Also you shouldn't need to manipulate the pins in this way. You should be able to use the `sjsu::lpc17xx::Pin` to control the pin configuration.

kammce, 2019-08-20:
> > Something to bear in mind with the L2 project is that it actually implements L1 pwm.hpp. This driver should be able to be used in place of a L1 peripheral pwm if the developer desires.
> 
> @kammce is the relationship to L1's pwm.hpp to control brightness from the seven segment display, or is this an intention towards a generalization for PCA9535s that would be off-board?

What I meant about this at the time was that the `PCA9535` driver actually implements `sjsu::Gpio`. But after working with Nick Derry on this we found that it is better for PCA pwm chip to do its own thing (not have an interface for now but we may add it later), but also have a method in the class that returns instances of `sjsu::Gpio`. Something like:

``` C++
Pca9535 io_expander;
io_expander.Initialize();
// Returns a Gpio implementation that communicates with the Pca9535 
// chip directly over i2c. This means you can pass the pins of the 
// Pca9535 various different external devices. 
sjsu::Gpio fault_led = io_expander.GetGpio(1); 
io_expander.SetAll(Gpio::State::kLow);
```

There is also the idea of making `Pca9535` implement the `ParallelBus` interface so it can act like a `Address Bus`, `Data Bus`, `Control Bus`, etc. But then again you could also just simply do this to make a parallel port:

``` C++
Pca9535 io_expander;
io_expander.Initialize();
GpioParallelBus control_bus = {
  io_expander.GetGpio(0),
  io_expander.GetGpio(1),
  io_expander.GetGpio(3),
  // Out of order because this is a good example of a scenario where we messed up on the board 
  // layout and swapped a few pins. Well its a easy fix with this. And now you don't have to do some 
  // irrating bit manipulation to make sure that your control signals go to the right place.
  io_expander.GetGpio(2),
};
```

And now you get an idea of what I invision about how this L2 module will be used.

ntwong0, 2019-08-22:
With 7cd3eb2 we are pretty much done with implementation and demo, so we just need to 
1. revise https://docs.google.com/document/d/1lpIfEZKWWlE4MQ8hM7hnPmjZ7m6DVhIw0dsD19KsgRY/edit#heading=h.akafqwf1ay8r
2. complete unit testing

ntwong0, 2019-10-04:
I see 47a1bedc71ce3e9d94577bd7029e72c4a918b599 and will update PCA9535 to use it.