libhal/SJSU-Dev2 #574: Fix run-test error LeakSanitizer & ptrace

ZackeryPlovanic, 2019-06-29:
When running a make run-test, LeakSanitizer encounters an error and fails the test. Running Ubuntu 18.04 in WSL. 

[image]

Text dump of above image:
```
==156==AddressSanitizer: failed to intercept 'puts'
==156==AddressSanitizer: failed to intercept 'scanf'
==156==AddressSanitizer: failed to intercept 'printf'
==156==AddressSanitizer: failed to intercept 'sprintf'
==156==AddressSanitizer: failed to intercept 'snprintf'
==156==AddressSanitizer: failed to intercept 'vsnprintf'
==156==AddressSanitizer: failed to intercept '_exit'
==156==AddressSanitizer: libc interceptors initialized
|| `[0x10007fff8000, 0x7fffffffffff]` || HighMem    ||
|| `[0x02008fff7000, 0x10007fff7fff]` || HighShadow ||
|| `[0x00008fff7000, 0x02008fff6fff]` || ShadowGap  ||
|| `[0x00007fff8000, 0x00008fff6fff]` || LowShadow  ||
|| `[0x000000000000, 0x00007fff7fff]` || LowMem     ||
MemToShadow(shadow): 0x00008fff7000 0x000091ff6dff 0x004091ff6e00 0x02008fff6fff
redzone=16
max_redzone=2048
quarantine_size_mb=256M
thread_local_quarantine_size_kb=1024K
malloc_context_size=30
SHADOW_SCALE: 3
SHADOW_GRANULARITY: 8
SHADOW_OFFSET: 0x7fff8000
==156==Installed the sigaction for signal 11
==156==Installed the sigaction for signal 7
==156==Installed the sigaction for signal 8
==156==T0: stack [0x7fffca3e2000,0x7fffcabe2000) size 0x800000; local=0x7fffcabe0f84
==156==AddressSanitizer Init done
===============================================================================
All tests passed (514 assertions in 38 test cases)

==157==Could not attach to thread 156 (errno 1).
==157==Failed suspending threads.
==156==LeakSanitizer has encountered a fatal error.
==156==HINT: For debugging, try setting environment variable LSAN_OPTIONS=verbosity=1:log_threads=1
==156==HINT: LeakSanitizer does not work under ptrace (strace, gdb, etc)
/home/zack/SJSU-Dev2/makefile:471: recipe for target 'run-test' failed
make: *** [run-test] Error 1
```

kammce, 2019-08-08:
Is this still a problem?

ZackeryPlovanic, 2019-08-09:
Yes the error is still present on my system

kammce, 2019-08-09:
@ZackeryPlovanic would you be able to try and figure out why this is happening as I cannot reproduce this on my machine.

kammce, 2019-10-09:
@MasterBoyd Can you check that this problem still exists for windows users when performing `make test` or `make library-test`.

ZackeryPlovanic, 2019-12-03:
Finally got around to running ./setup again and this error no longer persists on my machine.

closed 2019-12-03 by ZackeryPlovanic

kammce, 2019-12-03:
Woot! Thanks @ZackeryPlovanic!