libhal/SJSU-Dev2 #991: LTC4150 destructor should dettach gpio interrupt

kammce, 2019-12-16:


kammce, 2019-12-16:
@evanugarte Assigning this to you!

kammce, 2019-12-16:
An alternative to this would be to utilize the HardwareCounter peripheral which will handle detaching the  gpio itself.

evanugarte, 2019-12-23:
I wanted to abstract the `HardwareCounter` in `LTC4150` like so:
```cpp
  explicit constexpr Ltc4150(Gpio & int_pin,
                             Gpio & pol,
                             units::impedance::ohm_t resistance)
      : int_pin_(int_pin),
        pol_pin_(pol),
        resistance_(resistance),
        counter_(int_pin_, Gpio::Edge::kEdgeFalling), // instantiate hardwarecounter within the constructor
        pulses_(0),
        polarity_(Polarity::kDischarging)
  {
  }
```
this is proving to be difficult as I cannot find mock a constructor using FakeIt. Should I just instantiate the Hardware counter outside and make the constructor like below?

```cpp
explicit constexpr Ltc4150(HardwareCounter & int_pin,
                                          Gpio & pol,
                                          units::impedance::ohm_t resistance)
```

kammce, 2019-12-23:
That can work, but I need to add the ability to count up and down. This is something that I've been working on, just haven't pushed it yet. I'll try and get it in.

PR #1024 (2020-01-10): Add HardwareCounter to LTC4150
Resolves #991
commit 3b59363a (2020-01-10): Add HardwareCounter to LTC4150

Resolves #991
closed 2020-01-10 by kammce