libhal/SJSU-Dev2 #1421: Add pyOCD linux device rules

kammce, 2020-12-31:
# Bug Report

## Describe the bug

On many systems, specifically Windows host systems with a Ubuntu or other Linux VM, the stlink v2's do not seem to work. Even with udev rules they seem to fail. But the pyOCD rules seems to make this work.

https://github.com/pyocd/pyOCD/tree/master/udev

## To Reproduce

1. On a windows 10 machine
2. Install SJSU-Dev2
3. Using an stlinkv2 with any microcontroller attempt to flash
4. Should fail with LIBUSB_ERROR

## Expected behavior

Should debug or program device

## Screenshots

None at the moment

## Setup Information

- Host OS: WIndows 10
- Platform: any
- Browser: n/a
- Commit hash: Any/ N/A

## Additional context

Add any other context about the problem here.

manasabhyankar, 2021-01-27:
If you use Zadig to install LibUSB, this problem goes away: https://zadig.akeo.ie/. I initially had this problem with my STLink V2 as well and this executable solved it. Essentially, it installs WinUSB, libusb-win32/libusb0.sys or libusbK (as quoted from the site) and makes things work.

kammce, 2021-01-27:
Oh cool. We should look at the license for this program and see if we can just throw this into the `tools/` directory for Windows Users to run on their machines if they see issues with flashing and jtag flashing. 

If the license works out, could you make a PR adding this and a note in the documentation to help windows users out with flashing.