libhal/SJSU-Dev2 #1440: STlink ID

ulysses10, 2021-02-03:
# Bug Report

## Describe the bug

When attempting to flash Super Bluepill stm32f10x with the STlink V2, flash is unsuccessful due to STlink used having a different ID.
```
Warn : UNEXPECTED idcode: 0x2ba01477
Error: expected 1 of 1: 0x1ba01477
```

## To Reproduce

SJSU-Dev2/demos/stm32f10x/super_bluepill_led_blinker was the project used for this flash.

## Expected behavior

## Screenshots

```
Ulyssess-MBP:super_bluepill_led_blinker mactr0llii$ make flash PLATFORM=stm32f10x OPTIMIZE=s JTAG=stlink
---------------------------------------------------------------------
PLATFORM = 'stm32f10x' OPTIMIZE = 's' JTAG = 'stlink'
---------------------------------------------------------------------
======================================================================

   Memory region:     Used Size  Region Size     % Used
           FLASH:       32312 B        64 KB     49.30%
            SRAM:        1676 B        20 KB      8.18%

Section Memory Usage
   text    data     bss     dec     hex filename
  32012     300    1372   33684    8394 build/stm32f10x/s/firmware.elf

Programming chip via JTAG/SWD...
GNU MCU Eclipse 64-bit Open On-Chip Debugger 0.10.0+dev-00352-gaa6c7e9b (2018-10-20-08:28)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
adapter speed: 100 kHz
Info : clock speed 100 kHz
Info : STLINK v2 JTAG v37 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.168250
Warn : UNEXPECTED idcode: 0x2ba01477
Error: expected 1 of 1: 0x1ba01477
in procedure 'program' 
in procedure 'init' called at file "embedded:startup.tcl", line 496
in procedure 'ocd_bouncer'
** OpenOCD init failed **
shutdown command invoked

make: *** [jtag-flash] Error 1
```

## Setup Information

- Host OS: macOS Catalina Version 10.15.7 
- Platform: stm32f10x
- Browser: n/a
- Commit hash: 6dc595e97d52e89c42835b66acb2a212d470726e
## Additional context

Flash successful, after suggestion to add 
```
set CPUTAPID 0x2ba01477
``` 
to library/L0_Platform/stm32f10x/stm32f10x.cfg

danyhm, 2023-11-16:
This is not a bug, The STM32 you have is a clone

VictorQueiroz, 2024-07-04:
@danyhm Does that mean it's worse than the original? Is the "clone" problem an MCU problem or a board problem?

danyhm, 2024-07-04:
@VictorQueiroz 
It's a MCU problem. The wafer silicon is different from the one STM designed. It's a clone where some features seem to be different or not present at all.

I tested an identical HEX file on 2 different Bluepill boards, one with original STM and one with clone MCU. The clone hanged at intervals cause of stack overflows

kammce, 2024-07-04:
This is good info to know. Thank you @danyhm :) 

This discussion looks resolved, I'll resolve this comment.

closed 2024-07-04 by kammce